yaacov/node-modbus-serial #329: trashRequestId behavior

lonesome-coder, 2020-03-16:
I've been using this module to talk to an Accuvim Energy Meter on my network. However, I've been getting "Unhandled rejection (<{"err":"timeout"}>, no stack trace)" this error ever so often during the communication. I've been able to trace this back to a possible trashRequestId behaviour problem through some google searches. Would appreciate any help in resolving.

yaacov, 2020-03-16:
Hi, thanks for the issue.
If you get more data or find a fix / workaround, please comment here,
I've added a "help wnated" label, maybe someone with the same problem already have a fix.

Feel free to update the title of the issue as you get more info to help people that want to help you.

lonesome-coder, 2020-03-16:
This is a section of the debug log:

DEBUG : DSLHQEnergyMeter 1 requests in request pipe.
DEBUG : DSLHQEnergyMeter ready waiting
DEBUG : DSLHQEnergyMeter Sending pdu to the socket.
DEBUG : DSLHQEnergyMeter set timeout of 300 ms on send to wait for data
DEBUG : DSLHQEnergyMeter Data flushed.
DEBUG : DSLHQEnergyMeter received data
DEBUG : DSLHQEnergyMeter MBAP extracted
DEBUG : DSLHQEnergyMeter PDU extracted
DEBUG : DSLHQEnergyMeter received data and clean timeout PDU: {"type":"Buffer","data":[3,4,63,117,60,216]}
DEBUG : DSLHQEnergyMeter handling read holding registers response.
DEBUG : DSLHQEnergyMeter on read holding registers got PDU: {"type":"Buffer","data":[3,4,63,117,60,216]} pdu.length:6
DEBUG : DSLHQEnergyMeter request: {"fc":3,"defer":{},"pdu":{"type":"Buffer","data":[3,64,58,0,2]},"timeout":{"_called":false,"_idleTimeout":-1,"_idlePrev":null,"_idleNext":null,"_idleStart":8170,"_onTimeout":null,"_repeat":null,"_destroyed":false}}
DEBUG : DSLHQEnergyMeter waiting ready
0.958
DEBUG : DSLHQEnergyMeter 1 requests in request pipe.
DEBUG : DSLHQEnergyMeter ready waiting
DEBUG : DSLHQEnergyMeter Sending pdu to the socket.
DEBUG : DSLHQEnergyMeter set timeout of 300 ms on send to wait for data
DEBUG : DSLHQEnergyMeter Data flushed.
ERROR : DSLHQEnergyMeter Request timed out after 0.3 sec
DEBUG : DSLHQEnergyMeter waiting ready
**Unhandled rejection (<{"err":"timeout"}>, no stack trace)**

yaacov, 2020-03-16:
hi, thanks for the info,
can you add the code that produced this debug output ?

lonesome-coder, 2020-03-16:
const http = require('http');
const node_modbus = require('node-modbus');
const hex2ascii = require('hex2ascii');

const BevClient =  node_modbus.client.tcp.complete({
  'host':'192.168.90.50', //Beverage plant IP
  'port':502,
  'unitId':1,
  'timeout':200,
  'autoreconnect': true,
  'reconnectTimeout':15000,
  'logLabel':'BeverageEnergyMeter',
  'logLevel':'debug',
  'logEnabled':false
});

const DslClient =  node_modbus.client.tcp.complete({
  'host':'172.18.4.72', //DSLHQ IP
  'port':502,
  'unitId':1,
  'timeout':300,
  'autoreconnect': true,
  'reconnectTimeout':15000,
  'logLabel':'DSLHQEnergyMeter',
  'logLevel':'debug',
  'logEnabled':true
});

const time_interval = 1000;

DslClient.connect();

DslClient.on('connect',function(){
  setInterval(function(){
    DslClient.readHoldingRegisters(0x403a,2).then(function(data){
      console.log(int2float(data.register[0],data.register[1]));
    })
  }, time_interval)
})

function int2float(highWord,lowWord) {
  var buf = new ArrayBuffer(4);
  var ints = new Uint16Array(buf);
  ints[1] = highWord;
  ints[0] = lowWord;
  var floats = new Float32Array(buf);
  var num = floats[0];
  return num.toFixed(3);
}

yaacov, 2020-03-16:
> const node_modbus = require('node-modbus');

You are using the wrong module :smile: 
Please look at the examples: https://github.com/yaacov/node-modbus-serial/tree/master/examples

lonesome-coder, 2020-03-16:
OH! AH! hmm. But, the node-modbus module is built off of this module and the jsmodbus module, so it still should work no?
I'm using the "modbus-serial" module now and it works fine. Thanks much.

closed 2020-03-16 by lonesome-coder