yaacov/node-modbus-serial #330: SyntaxError: Unexpected token ...

HughF, 2020-03-17:
I've tested this library with node 8.10.0 on x86 with no issues, but when I install it from npm on an arm target running 6.17.0 I get the following error:

`debian@Vigo:~$ node modbusTester.js 
/home/debian/node_modules/modbus-serial/node_modules/@serialport/stream/lib/index.js:103
  const settings = { ...defaultSettings, ...options }
                     ^^^

SyntaxError: Unexpected token ...
    at createScript (vm.js:56:10)
    at Object.runInThisContext (vm.js:97:10)
    at Module._compile (module.js:549:28)
    at Object.Module._extensions..js (module.js:586:10)
    at Module.load (module.js:494:32)
    at tryModuleLoad (module.js:453:12)
    at Function.Module._load (module.js:445:3)
    at Module.require (module.js:504:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/debian/node_modules/modbus-serial/node_modules/serialport/lib/index.js:1:82)
`
This target already had a working install of the serialport library that I was using to communicate with some other ports on the target. Is this a problem?

yaacov, 2020-03-17:
Did you try the version without included serialport ?
https://www.npmjs.com/package/modbus-serial/v/7.8.1-no-serial-port

HughF, 2020-03-17:
Sorry, I did not know there was a version without serialport. I did and unistall of the previous version, then re-installed the recommended version - everything seems ok now, just a small error due to port locking that I will fix. Thanks

yaacov, 2020-03-17:
np. have fun.

we should document it in a more noticeable place, It will be nice if you can look at the README.md file, and suggest in a pull request some notes about the no-serialport option, and when to use it.

HughF, 2020-03-17:
OK, I will see if I can get time to add notes about the use of this version.

closed 2020-03-17 by yaacov