yaacov/node-modbus-serial #331: How to write data to a holding register?

bluemiaomiao, 2020-03-24:
I want to write the number 0 to the holding register.

yaacov, 2020-03-24:
hi, thanks for the question :green_heart: 

what did you try, can you post the code ?
what error / unexpected behaviour do you get ?

bluemiaomiao, 2020-03-24:
I have not encountered any mistakes.
I read the document and found no way to write data to the register, only to send a request and read the data. I want to write the data I have prepared to the register.

yaacov, 2020-03-24:
did you try:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/write.js

bluemiaomiao, 2020-03-24:
the code:
```javascript
client.writeRegister(40001, 1);
```

the error:
```
Uncaught (in promise) Error: Modbus exception 2: Illegal data address (register not supported by device)
    at RTUBufferedPort.<anonymous> (D:\modbus-client\node_modules\modbus-serial\index.js:298)
    at emitOne (events.js:116)
    at RTUBufferedPort.emit (events.js:211)
    at RTUBufferedPort._emitData (D:\modbus-client\node_modules\modbus-serial\ports\rtubufferedport.js:108)
    at SerialPort.onData (D:\modbus-client\node_modules\modbus-serial\ports\rtubufferedport.js:71)
    at emitOne (events.js:116)
    at SerialPort.emit (events.js:211)
    at addChunk (_stream_readable.js:263)
    at readableAddChunk (_stream_readable.js:250)
    at SerialPort.Readable.push (_stream_readable.js:208)
```
I want to write the number 1 to the holding register at 40001

yaacov, 2020-03-24:
> Illegal data address (register not supported by device)

I looks like your device does not support this address.

are you sure it's really 40001 ?
some documentations adds 40000 to denote holding registes, so it they mean *holding register* number 1, they will write *register* 40001. 
you should also note that the first register may be 0 or 1, so you may be looking for register 0 instead of register 1.

bluemiaomiao, 2020-03-24:
```javascript
client.readHoldingRegisters(0, 8)
```
This is the way I read data to the holding register. I want to write data to this.

bluemiaomiao, 2020-03-24:
my data:
```javascript
client.readHoldingRegisters(0, 8).then((response) => {let mydata = response.data[1]});
```
This is the way I read data to the holding register. I want to write data to this. How to do it?

yaacov, 2020-03-24:
for example, starting at address 0: 
https://github.com/yaacov/node-modbus-serial/blob/master/examples/write.js#L40
``` js
client.writeRegisters(0, [10, 9, 8, -20 + 65535, -10 + 65535]);
```

yaacov, 2020-03-25:
@bluemiaomiao did the example help you ?

bluemiaomiao, 2020-03-28:
:) thx. @yaacov

closed 2020-03-28 by yaacov