yaacov/node-modbus-serial #332: timeout

JianJD, 2020-04-01:
I call the writefc3 function and get no result from the device. I only got a timeout result, but the device was OK. Send the same command and I can get the correct result with other software. I see that your source code directly calls the  timeout method

`function _writeBufferToPort(buffer, transactionId) {
    var transaction = this._transactions[transactionId];
    
    this._port.write(buffer);
    
    console.log(transaction,'transaction')
    if (transaction) {
        transaction._timeoutFired = false;
        transaction._timeoutHandle = _startTimeout(this._timeout, transaction);
    }
}`

yaacov, 2020-04-01:
Hi, thanks for the question.

You can try setting the timeout:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/simple.js#L28

JianJD, 2020-04-01:
When I set the timeout, I get the timeout result. Without setting the timeout, I get no result. When is the open method called

JianJD, 2020-04-01:
This is my code
`var client = new ModbusRTU();
client.connectTCP("10.100.0.72", { port: 4001 }).then(async res=>{
        client.setID(2)
        client.setTimeout(1000);
        client.writeFC3(2,3,3,(data)=>{
            console.log(data,'callbackData')
        })
     }).catch(r=>{
         console.log(r)
         return false
     })`

yaacov, 2020-04-01:
> When is the open method called

the `client.connect[TYPE]` methods are wrappers of the `open` method, the `open` method is called by the 'connect...', you can use `open` or `connect...`

JianJD, 2020-04-01:
Do I want to get the data returned by the device using the open method or the connect method?

yaacov, 2020-04-01:
They do the same thing ... it's what ever looks nicer to you.
In all the examples I usually, use the promise versions e.g. `connect...` and `read...` but
I see that you use `writeFC3` which is from the none promise, so I do not see benefits to you for 
using one or the other.

Xsandor, 2022-01-13:
@yaacov, the data is returned in the second parameter of the callback function, but he only used a single parameter, so he would only get the possibly empty error, right?

yaacov, 2022-01-13:
stale issue, closing

closed 2022-01-13 by yaacov