yaacov/node-modbus-serial #333: question: How can I emulate Modbus Exception 2/3 using ServerTCP

robertsLando, 2020-04-15:
Hi, I would like to try emulate this exceptions with `ModbusRTU.ServerTCP` but If I try to throw an error in vectors methods there is no way to catch it. 

How can I manage this? Thanks

yaacov, 2020-04-15:
Hi, Thanks for the issue :green_heart: 

> but If I try to throw an error in vectors methods there is no way to catch it

:+1: feel free to make a pull request fixing it.

> Also if I return an empty array when the read should fail the error is passed to callback

callback should be optional ... so it's nice to have a fall-back if callback is undefined ...

> should that throw statement be removed?

hmm, in some places we do not have it ( e.g. https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp_handler.js#L204 ) while in others we do ( e.g. https://github.com/yaacov/node-modbus-serial/blob/master/servers/servertcp_handler.js#L280) :-)

I think it's safe to remove it in all places, or check if callback exist and to the right thing :-)

robertsLando, 2020-04-16:
I think I will go using vectors with callbacks as seems that this should handle the error If I return 

```
{
            modbusErrorCode: 0x02, // Illegal address
            msg: "Invalid length"
}
```

> I think it's safe to remove it in all places

I think too, will submit a PR for that

PR #334 (2020-04-16): fix: Remove error throw from servertcp_handlers
Fixes #333
commit c3799018 (2020-04-16): fix: Remove error throw from servertcp_handlers (#334)

Fixes #333
closed 2020-04-16 by yaacov