yaacov/node-modbus-serial #335: bug: Sometimes close callback isn't called (Modbus TCP)

robertsLando, 2020-04-16:
I'm writing some test suites of my package and I have this bug, if I inpect modbus-serial I can see that 
[here](https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L145) the callback is assigned correctly but when I reach the client close event [here](https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L96) and `handleCallback` is called callback is null. I'm unable to understand what makes it null. 

What I'd doing in test is: 

- Open the connection
- Close the connection
- Open the connection
- Close the connection <--- here stucks

@yaacov Any ideas?

robertsLando, 2020-04-16:
@yaacov I'm happy to help here but have you any hint?

yaacov, 2020-04-16:
> @yaacov I'm happy to help here but have you any hint?

:laughing: 

Can't look at it now ... will look at it later and have an answer :-)

robertsLando, 2020-04-16:
@yaacov Ok thanks :) I'm still debugging it anyway, If I find a solution I will let you know

robertsLando, 2020-04-16:
More info: seems a little step is done If I use `setImmediate` to call the close method. 

To be more clear on what is going on here, In my client library I emit `connected` event in the callback function of `Modbus.open`, here the first time I don't do anything but on the second `connected` I close the client, here if I use `setImmediate` the callback returns as it should. That's weird but I think that it's someting related to javascript functions stack... 

But even with this method I get some errors

yaacov, 2022-06-01:
@everhardt hi, thank you for noticing it,
can you try to find the issue and fix it ?