yaacov/node-modbus-serial #337: Support for Extended Addresses?

mrarmyant, 2020-05-03:
Trying to write some device emulators with this awesome package.  Works great except for the fact some of the holding registers i need to work with are 3000002-300026.  When I try this with a client and server both using node-modbus-serial I get an error "The value of "value" is out of range. It must be >= 0 and <= 65535. Received 300002"

Is there a way to let it use extended address values?

yaacov, 2020-05-04:
> Is there a way to let it use extended address values?

This modbus function use 2 bytes for addressing, so range is 0..65535 .

You can not use addresses above 65535,
If you want to use address space above 65535, you will need to map it on both ends to and from the available range..

khosengman, 2020-07-16:
>  Is there a way to let it use extended address values?
> 
> This modbus function use 2 bytes for addressing, so range is 0..65535 .
> 
> You can not use addresses above 65535,
> If you want to use address space above 65535, you will need to map it on both ends to and from the available range..

May u show some example what is the meaning of "need to map it on both ends to and from the available range" ?

khosengman, 2020-07-16:
> > Is there a way to let it use extended address values?
> 
> This modbus function use 2 bytes for addressing, so range is 0..65535 .
> 
> You can not use addresses above 65535,
> If you want to use address space above 65535, you will need to map it on both ends to and from the available range..

May u show some example what is the meaning of "need to map it on both ends to and from the available range" ?

khotbs, 2020-07-16:
> > Is there a way to let it use extended address values?
> 
> This modbus function use 2 bytes for addressing, so range is 0..65535 .
> 
> You can not use addresses above 65535,
> If you want to use address space above 65535, you will need to map it on both ends to and from the available range..

May you provide some example how to perform this ? " map it on both ends to and from the available range "

sigma-design, 2020-07-16:
Can it be just a typical (PLC) use case where 1st digit is just signifying type of command while the rest is the actual address ?
Below is the excerpt from Wiki on Modbus:

Coil, discrete input, input register, holding register numbers and addresses

Some conventions govern how access to Modbus entities (coils, discrete inputs, input registers, holding registers) are referenced.

It is important to make a distinction between entity number and entity address:

    Entity numbers combine entity type and entity location within their description table.
    Entity address is the starting address, a 16-bit value in the data part of the Modbus frame. As such its range goes from 0 to 65,535

In the traditional standard[citation needed], numbers for those entities start with a digit, followed by a number of 4 digits in the range 1–9,999:

    coils numbers start with 0 and span from 00001 to 09999,
    discrete input numbers start with 1 and span from 10001 to 19999,
    input register numbers start with 3 and span from 30001 to 39999,
    holding register numbers start with 4 and span from 40001 to 49999.

This translates into addresses between 0 and 9,998 in data frames. For example, in order to read holding registers starting at number 40001, corresponding address in the data frame will be 0 with a function code of 3 (as seen above). For holding registers starting at number 40100, address will be 99. Etc.

This limits the number of addresses to 9,999 for each entity. A de facto referencing extends this to the maximum of 65,536.[14] It simply consists of adding one digit to the previous list:

    coil numbers span from 000001 to 065536,
    discrete input numbers span from 100001 to 165536,
    input register numbers span from 300001 to 365536,
    holding register numbers span from 400001 to 465536.

sigma-design, 2020-07-16:
So, in your case of 300026 it may mean "input register number 26" (with address 25)

greggwon, 2020-08-18:
In many modbus device documents of old, addresses on the device start with '0' for coils, '1' for status, '3' for input and '4' for input registers.  These addresses are typically '1' based and that means that 40001 is the first holding register.   Since modbus uses 'offsets' for a register type in this case, the actual coding of such a request would use the value 0 for the first register.  Thus, anytime you see an address expressed as a number starting with one of the above four digits with 5 or more digits specified, it should most likely be handled the same way.   Thus, 400020 is the same as 40020.  Technically, even 4000020 is also the same register, even though you can't ever have more than 465536 as the largest holding register address, some documentation will include a 0 dividing character, always so that you may see 4065536 so that it's easy to see that this is a register type and address expression and not a register offset value that you would put into your modbus request packet.    So, remove the 40, subtract 1, and put 0xffff in as the offset value in this case.  For the 400020 case, the offset is 20-1 or 0x0013.

larson-zss, 2020-09-17:
@yaacov  hello,I'm reporting the same mistake,"The value of "value" is out of range. It must be >= 0 and <= 65535. Received -25",What if the instruction I need to issue is negative?Looking forward to your reply

Xsandor, 2022-01-13:
@yaacov, I believe you can close this issue.

The explanation from greggwon is correct and should solve the problem for anyone who encounters extended addressing.

@larson-zss, a negative register address is not valid, must be a mistake in the documentation for the device you are trying to communicate with.

yaacov, 2022-01-13:
Thanks all

closed 2022-01-13 by yaacov