yaacov/node-modbus-serial #341: Upgrade to SerialPort v9.0.0 required

SaikatCool, 2020-06-07:
Hello,

Can this library be updated to use SerialPort v9.0.0.

I am trying to use node-modbus-serial in a ElectronJS app and running into issues.

According to [this](https://github.com/serialport/node-serialport/issues/2031#issuecomment-626324067), the fix is available in SerialPort v9.0.0.

yaacov, 2020-06-07:
Hello, thanks tor the issue.

Sure, can you make a pull request bumping the version ?

SaikatCool, 2020-06-07:
@yaacov , pull request on the way.
Only package.json changes fine right?

SaikatCool, 2020-06-07:
@yaacov , I am not able to push the changes in a new branch. It says access denied.

[image]

SaikatCool, 2020-06-07:
@yaacov  kindly help. I need to get this fixed as soon as possible. Have hit a roadblock :(

yaacov, 2020-06-07:
to create a pull request on a project you do not have write permission on you need to open a fork:
https://guides.github.com/activities/forking/

once you have a fork you can create a pull request:
https://guides.github.com/activities/forking/#making-a-pull-request

have fun :white_flower:

SaikatCool, 2020-06-07:
@yaacov thanks for the tip. Still new to using GitHub.

https://github.com/yaacov/node-modbus-serial/pull/342

Could you please check?

SaikatCool, 2020-06-07:
@yaacov  thanks for approving the pull request.
When the v7.8.2 be published to NPM? so that I can consume it in my project and test.

yaacov, 2020-06-07:
thanks, merged :green_heart:

yaacov, 2020-06-07:
https://www.npmjs.com/package/modbus-serial/v/8.0.0

SaikatCool, 2020-06-07:
Thanks, will test and give you a feedback, @yaacov

SaikatCool, 2020-06-07:
It works!

Thanks for your good work. Happy to contribute :)

SaikatCool, 2020-06-07:
@yaacov  could you also please create a tag for v8.0.0? thanks in advance.

closed 2020-06-07 by SaikatCool

yaacov, 2020-06-07:
sure, have fun