yaacov/node-modbus-serial #344: Accessing the request and response PDUs?

sjlongland, 2020-06-15:
Hi,

This is a bit of a silly question, we're developing an application built on `modbus-serial` and a request has been made that for debugging purposes, the raw PDUs sent to the Modbus device be exposed as hex dumps for debugging purposes.

Right now, if I do a `readHoldingRegisters` I get back something like:
```
{
    "data": [… array of ints or bools …],
    "buffer": Buffer object
}
```

The `buffer` parameter is fantastic for extracting lots of data you can't otherwise do (e.g. floating-point registers).  However, if there's a CRC error, some of the engineers I'm working with would like to see a hex dump of the entire response including Modbus PDU header and CRC.  Likewise, they'd also like to see the request PDU that was sent to the meter as well.

I recognise this will "differ" depending on whether it is Modbus/RTU, Modbus/ASCII or Modbus/TCP.  I'm thinking something akin to an extension of existing return value data:

```
{
     /* as you have now */
    "data": [… array of ints or bools …],
    "buffer": Buffer object,
    /* new data */
    "request": Buffer object,
    "response": Buffer object
}
```

Likewise, for exceptions, maybe `request` and `response` properties could be added to the `Error` instance before it is thrown.

Is there a way to achieve this with `modbus-serial` as it stands now, or do I need to prepare a pull request adding this functionality?

yaacov, 2020-06-15:
Hi, thank you for tor the issue

> Is there a way to achieve this with modbus-serial as it stands now, or do I need to prepare a pull request adding this functionality?

You will need to prepare a pull request to add the functionality you need.

sjlongland, 2020-06-22:
No problems, this is a work-in-progress, but having a look, I see the Modbus port infrastructure abstracts the underlying transport to make it appear like Modbus/RTU, so I've sort of just used that.

So the payloads provided are Modbus/RTU, even if the underlying protocol is Modbus/TCP or Modbus/ASCII, as that seems to be the less invasive way to do it.  (9/10 times for us, it _will_ be Modbus/RTU over the wire anyway via a Modbus/RTU-Modbus/TCP gateway.)

https://github.com/widesky/node-modbus-serial/tree/feature/WC-2924-add-debug-data

If the user sets `isDebugEnabled` on the Modbus client object; they receive two new properties in responses:
* `request` is the request PDU, as a `Buffer`
* `responses` is the `Array` of all responses seen including bad CRCs, etc.  All of these are `Buffer`s too.

Likewise, on exceptions; `modbusRequest` and `modbusResponses` are added.  If colleagues are happy with the caveat about it always being Modbus/RTU format, then I'll submit the pull request.

sjlongland, 2020-06-25:
Resolved following merge of PR#349.  Many thanks. :-)

closed 2020-06-25 by sjlongland

yaacov, 2020-06-25:
np :green_heart: 

thank you for the patch :+1: