yaacov/node-modbus-serial #348: error handling, long holding time after catching an error?

g-zajac, 2020-06-24:
I use the function below to connect, read registers and disconnect from a modbus device:
```javascript
async function readRegisters(ip, registerAddress, numberOfRegisters){
  try {
    client.setID(1);
    client.setTimeout(1000);
    await client.connectTCP(ip, {port: 502})
    const regs =  await client.readHoldingRegisters(registerAddress, numberOfRegisters);
    return regs;
  }
  catch (e) {
    return e
  }
  finally {
    client.close();
  }
}
```

and run it:
```javascript
readRegisters("192.168.1.10", 1, 5)
.then((readings) => {console.log("readings:", readings);})
.catch((error) => {console.log("error: ", error);})
```

When I run it with a modbus device connected it works fine. When the device is disconnected it outputs an error but then hangs for long time, like a minute or so till function finishes. How can I reduce this time?

yaacov, 2020-06-25:
Hi, thank you for the issue.

their was a change 2 days ago in the code that handle that:
https://github.com/yaacov/node-modbus-serial/pull/347

can you use the code from github (not npm) and see if current development code fixes your issue ?

g-zajac, 2020-06-25:
Hi, thank you for prompt answer. I have uninstalled the npm modbus-serial package and installed it again from git:
`"modbus-serial": "git+https://github.com/yaacov/node-modbus-serial.git"`
still the same I'm afraid. I use modbus for controlling [Moons SSDC10-D](https://www.moonsindustries.com/p/ssdc-series-step-servo-drives/ssdc10-d-000004696351003247) and stepper motors. It connects, reads registers and closes connection quickly with no issues. But if there is no device I got caught error message immediately and around 1min till disconnecting.
Any suggestions?

yaacov, 2020-06-26:
a - can you try to set the timeout in the potions ?
e.g. `await client.connectTCP(ip, {port: 502, timeout: 1000})`
b - can you try and remove the override in:
https://github.com/yaacov/node-modbus-serial/blob/master/apis/connection.js#L107
and see if it changes something ? (after you added the timeout in the options :-))

g-zajac, 2020-06-26:
I have added timeout to options:
```javascript
async function readRegisters(ip, registerAddress, numberOfRegisters){
  try {
    client.setID(1);
    client.setTimeout(999);
    await client.connectTCP(ip, {port: 502, timeout: 888})
    const regs =  await client.readHoldingRegisters(registerAddress, numberOfRegisters);
    return regs.data;
  }
  catch (e) {
    return "modbus connection error"
  }
  finally {
    client.close();
  }
}
```
I have log out from the apis/connection.js section if (this._timeout) {...
this._timeout = 999
options.timeout before overriding : 888
options.timeout after overriding : 999

Commenting out the if(this._timeout) section and adding timeout in options does not change the disconnection time, still long.

yaacov, 2020-06-26:
hmmm... so the next place to search is:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L137

can you set a timout after the open call ?
```
timer = setTimeout(function() {
            // check what is going on :-)
}, timeout);
```

we can printout the state of the socket, maybe force close it if not connection is done and we hit the timer / emit timeout signal / soemthing else ... ?

g-zajac, 2020-07-09:
Hi, I have added a setTimeout like this:
```js
async function readCoils(ip, coilAddress, numberOfCoils){
  try {
    setTimeout( () => {
      local_client.close;
    }, 900);

    var local_client = new ModbusRTU();
    local_client.setID(clientID);
    local_client.setTimeout(1000);
    await local_client.connectTCP(ip, {port: 502})
    const coils =  await local_client.readCoils(coilAddress, numberOfCoils);
    return coils.data;
  }
  catch (e) {
    return {connected: false}
  }
  finally {
    local_client.close();
  }
}
```
still the same, throws error:
`TCP Connection Timed Out {"stack":"Error: TCP Connection Timed Out\n    at Socket.<anonymous> (/server/node_modules/modbus-serial/ports/tcpport.js:11
1:24)\n    at Socket.emit (events.js:198:13)\n    at Socket._onTimeout (net.js:442:8)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}`
and hang about for a minute when modbus device disconnected, even when timeOut calls. Any suggestions how can I force connection close please?