yaacov/node-modbus-serial #350: Dealing with multiple TCP modbus devices connections?

g-zajac, 2020-07-07:
Hi
I'm trying to read registers from multiple devices at the same time with code:
```js
var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

const clientID = 1;
const timeOut = 1000;

function setClient() {
  client.setID(clientID);
  client.setTimeout(timeOut);
}

async function readRegisters(ip, registerAddress, numberOfRegisters){
  try {
    setClient();
    await client.connectTCP(ip, {port: 502})
    const regs =  await client.readHoldingRegisters(registerAddress, numberOfRegisters);
    return regs.data;
  }
  catch (e) {
    logger.error("modbus error", e);
    return e
  }
  finally {
    client.close();
  }
}
```
when I invoke readRegisters once it works fine, but when call it twice at the same time with different IPs it does not work:
```js
var interval = setInterval(function(){
   Promise.all([readRegisters("192.168.1.10", 0, 10), readRegisters("192.168.1.20", 0, 10)])
   .then((resp) =>{ console.log("resp", resp);})
   .catch((e) => {console.log("error", e);})
}, 1000)
```
throwing error:
`
Port Not Open {"name":"PortNotOpenError","errno":"ECONNREFUSED","stack":"PortNotOpenError: Port Not Open\n    at ModbusRTU.writeFC4
`

What would be the best approach, practise to handle multiple device connections please? I have got 15 modbus devices and would like to read registers of all of them a few times per sec. Any suggestions please? Thank you

yaacov, 2020-07-07:
Hi, thank you for the issu.

what happen if you create a connection, and then poll the same connection periodically without closing it ?

yaacov, 2020-07-07:
you need to finish using the `client` for one device before starting using it for the next.
solution can be:

create more them one client, and use different client for each device:
``` js
var client_for_device_1 = new ModbusRTU();
var client_for_device_2 = new ModbusRTU();
...
```

or to create a client per register read:
``` js
async function readRegisters(ip, registerAddress, numberOfRegisters){
   var local_client = new ModbusRTU();
   ...
```

or building the calls in a serial order, so one will finish before starting another

or ...

g-zajac, 2020-07-08:
client per read works great, thank you!

closed 2020-07-08 by g-zajac