yaacov/node-modbus-serial #352: Understanding [object Object] error, whether my socket is closing?

ceraz68, 2020-07-23:
I have a modbus TCP connection to a solar inverter that reported by an error "[object Object]". A relaunch of the code resolves the issue but would like to understand what is happening, so the code can be improved when this happens.

I was wondering if this error "[object Object]" is due to the socket being closed or my code is missing something. I don't want to reconnect if this will create loads of open sockets on the inverter for no good reason.

My code opens a connection any read registers every 5 min between 8am and 8pm. So the TCP connection is idle for a long time (but TCP connection stay open for long periods).

Thanks a lot for any help 👍 

Extract of my node.js code is very simple ... variables are defined in the code:

//Open connection to configured IPv4 address of SMA inverter
var sma_inverter = new ModbusRTU(); //code loads modbus-serial version 7.8.6
sma_inverter.setID(3);
sma_inverter.connectTCP(this.smaipaddress, {port: 502 });

//SetTimeout to poll readings from the inverter every 5 minutes
setInterval(function() {
   if (hour >= 8 && hour <= 20) {  // I only read during the day
   sma_inverter.readHoldingRegisters(30775, 10)
 	.then (function(data) {
         self.power = data.buffer.readUInt32BE();
         if(self.power > 999999 || self.power == 0) self.power = 0.0001; //forced value for later processing
         console.log(self.power);
         })
    	.catch (function(e) {
	  console.log("SMA read error: " + e); //This reports [object Object] errors
          });
    };
}, self.refreshInterval); //refresh interval is 5 minutes

Malvineous, 2020-08-30:
You're converting the error object into a string, you need to see the whole object instead: `console.log(e);`

yaacov, 2020-08-30:
Thank you for the issue :green_heart: 

Can you make a pull request fixing that ?

Malvineous, 2020-08-30:
It's not a problem with the library!

ceraz68, 2020-08-31:
Ah ok i see what I did, by appending the quotes the error gets converted to a string. Thanks a lot for looking at this, I have limited coding skills.

closed 2020-08-31 by ceraz68