yaacov/node-modbus-serial #356: Allow receive data length bigger than expected

E-NINA, 2020-09-01:
Hello,

I've asked on the repo https://github.com/BiancoRoyal/node-red-contrib-modbus about an issue with a communication with my company energy meter.
But it seems that issue come directly from your base library :
I will try explain what is the issue, I'm getting problem with a data length bigger than expected "Error: Data length error, expected XX got XX"
Peer the manual of meter there are single and double registers all on consecutive addresses:

Index       Address            data
1              0x0001             FF FF (single)
2              0x0002             FF FF (single)
3              0x0003             FF FF (single)
...
38              0x0026           FF FF FF FF (double)
39              0x0027           FF FF FF FF (double)
40              0x0028           FF FF FF FF (double)
...
108            0x006C           FF FF (single)
109            0x006D           FF FF (single)
...

When I read single registers like 1 or 108 I've no problem
I get "Error: Data length error, expected 7 got 9" when I try read "double registers"

Meter information JANZ 3801 (my national used company meter)

**Example of single register:**
Voltage -> Index 108 (HEX -> 6C )
resquest: 00 01 00 00 00 06 01 04 00 6c 00 01
response: 00 01 00 00 00 05 01 04 02 **_08 E5_** -> (hex 08 e5 -> 2277 -> 227,7V)

**Example of double register:**
rate 1 total energy -> Index 38 (HEX -> 26 )
resquest: 00 01 00 00 00 06 01 04 00 26 00 01
response: 00 01 00 00 00 07 01 04 04 **00 01 92 B1** -> (hex 00 01 92 B1 -> 103089 -> 103,089 kwh)

If I try pull 2 consecutive data registers I get, similar error message, is them:

_"Error: Data length error, expected 9 got 13"_

Replay data is:

resquest: 00 01 00 00 00 06 01 04 00 26 00 02
response: 00 01 00 00 00 0B 01 04 08 **00 01 A2 CA, 00, 00, 58, 58**

00 01 A2 CA are the data for register 0x26
-> rate 1 total energy - 107,210 kwh

00, 00, 58, 58 are the data for register 0x27
-> rate 2 total energy - 22,616 kwh

All examples above are from a windows program that  can communicate with my device and no problem at all (data/examples I provide are taken from that), issue seems is with library, that brake with the messages "Data length error, expected XX got YY" and no data are show/retained.
Is there a way to fix this on library, like add a switch to not compare data lengths, or a function to mention expected size of response?
If not, can give me some hints how to delete/baypass this check on the code of plugin? (I'm not a programmer so please be patient with me)

Best Regards
ENina

yaacov, 2020-09-01:
hi, thank you for your issue.

you may want to check out: https://github.com/yaacov/node-modbus-serial/blob/781ea7e64e52aa2d94feae96cceaf3caa1955708/index.js#L321
try to disable the check, and see if it solves your problem.

E-NINA, 2020-09-01:
Hello,

Bypassing check on lines 321 till 326 didn't make any effect (I just comment them, error was the same)
I see them similar check on lines 355 ... 364  (**_check message length if we do not expect this data raise an error_**)
I commented those lines and put back in the code lines 321 till 326 and that did the effect.
Now I can see data coming and seems is running normal.

So for me I can work around with this situation, but for other users you think it will make sense add some kind of option to bypass this test? I think will be useful for other situations.

Anyway, many thanks for your support.

Best Regards
ENina

yaacov, 2020-09-01:
> So for me I can work around with this situation, but for other users you think it will make sense add some kind of option to bypass this test? I think will be useful for other situations.

I don't know. what will happen if we disable this check completely ?

> Anyway, many thanks for your support.

np :-) if you think of a nice solution and want to submit a pull request that fixes that, feel free :-)