yaacov/node-modbus-serial #357: custom request

bechterev, 2020-09-01:
How can I create a custom query or access functions that are not included in the examples? For example, FC 20 - Read file record.

yaacov, 2020-09-01:
hi, thanks for the question, you can take a look here https://github.com/yaacov/node-modbus-serial/pull/319

bechterev, 2020-09-01:
I don't quite understand, are you suggesting using FC43/14 instead of FC20 (0x14) or have you added it to the list of future features?

yaacov, 2020-09-01:
> I don't quite understand, are you suggesting using FC43/14 instead of FC20 (0x14)

no, but you can look at how function code 34 was added and learn how to make a pull request adding support for new function code, then you could make your pull request.

bechterev, 2020-09-01:
Okay, I'll try.

bechterev, 2020-09-02:
Не могу понять почему параметр next все время undefined, как его правильно задать. Зацикливается из-за этого параметра. Делаю так
I can't understand why the next parameter is always undefined, and how to set it correctly. Looped because of this parameter. Do so
ModbusRTU.prototype.writeFC20 = function(address, records, length, **next**){

bechterev, 2020-09-02:
I can't understand why the **next** parameter is always undefined, and how to set it correctly. Looped because of this parameter. Do so
ModbusRTU.prototype.writeFC20 = function(address, records, length, **next**){

yaacov, 2020-09-02:
`next` is a function that is called when the operation is finished, if it's undefined, no function is called operation is done, if it's a function, the function will be called with relevant parameters

bechterev, 2020-09-02:
use let f=client.readFileRecords(249, 10).then(x=>console.log(x)).catch(err=>console.log(err));

bechterev, 2020-09-02:
tell me how to do next not undefined

bechterev, 2020-09-02:
sorry, I figured it out

bechterev, 2020-09-03:
I added a function to read the file, but at the end of reading I get the error "illegal data address". How do I get standard error 94?

bechterev, 2020-09-04:
pr #359

closed 2020-09-09 by bechterev