yaacov/node-modbus-serial #362: Reading Multiple Modbus RTU with different SerialPort settings

komorebi-san, 2020-09-10:
Hi,

Is there a way to correctly read multiple slave devices in series, with different baudRate and parity?

For example:

Device ID: 10
Baurate: 9600
Parity : None
Stopbits: 1

Device ID: 15
Baurate: 19200
Parity : Even
Stopbits: 0

Device ID: 17
Baurate: 9600
Parity : Even
Stopbits: 1

So the program will loop through each config and set the serial port according to the setting.

yaacov, 2020-09-10:
hi, thank you for your question.

> Is there a way to correctly

No, using multiple baudrates on the same physical line is not recommended.
But if If it works for you it is correct for you :-)

My first thought is to close last one and open new client for each call ... but you may find a better way.

p.s.
If you find a good way to do this, please consider creating a pull request adding and example for using one serial port in multiple baudRates, to the examples directory.

komorebi-san, 2020-09-10:
@yaacov 

My first thought as well to open and close repeatedly. 

However, the response time of opening and closing the serialport is not ideal. Let me do more research and let you know if I find a better way of doing it. 

Thanks!

closed 2020-09-10 by komorebi-san