yaacov/node-modbus-serial #363: Issue a resolution with a negative value

larson-zss, 2020-09-18:
I had a problem. I'm working on a treadmill project, now meet a problem, The treadmill need to reverse, instructions for The issuance of The negative, but The correct plug-in 'The value of "value" is out of range. It must be > = 0 and < = 65535', I also tried to figure into two's complement, but also can not, I don't know how to deal with this problem now.If you know how to solve it, please contact me, thank you

yaacov, 2020-09-18:
hi, thank you for the issue.

Modbus protocol does not specify how to send negative values, users can choose what suite their use case, the protocol to send negative values is a specification of the machine you are using, you should consult machine documentation, or the machine manufacturer.

larson-zss, 2020-09-18:
Ok, I see. Thank you for your reply

625394976@qq.com
 
From: Yaacov Zamir
Date: 2020-09-18 15:58
To: yaacov/node-modbus-serial
CC: larson-zss; Author
Subject: Re: [yaacov/node-modbus-serial] Issue a resolution with a negative value (#363)
hi, thank you for the issue.
Modbus protocol does not specify how to send negative values, users can choose what suite their use case, the protocol to send negative values is a specification of the machine you are using, you should consult machine documentation, or the machine manufacturer.

larson-zss, 2020-09-21:
The problem is solved. I first convert the negative number into binary, and then do the reverse code and complement operation, get the binary data and then convert to decimal, issue, problem solved

closed 2020-09-21 by larson-zss