yaacov/node-modbus-serial #364: [question] How to scan a modbus network

robertsLando, 2020-10-02:
Hi @yaacov , firstly thanks again for this awesome module :)

I'm looking for a way to scan available unitsIds in a modbus network, I was thinking about simply try to read for example the holding register 1/0 and if the function returns I can say there is an available device matching that unitid, but I wonder if there is a better way to implement this. 

Any suggestions?

yaacov, 2020-10-03:
Hi, thank you for the question.

> Any suggestions?

As you suggested, pooling a ** known register you know exist in all your devices ** sound a good option.

p.s.
When implementing you should think of the affect such a poll cycle will have on other readings, on a serial line, pooling all addresses may take some time.

p.p.s. 
IMHO it depends on the devices you have,
some devices may have a status register that is specifically designed so you can poll it at known intervals and keep a table of connected devices and their status.
some devices don't like to be pooled and may cause delays and noise if pooled to many times or try to contact bad addresses.

robertsLando, 2020-10-03:
I heard about a function code 17 that may be used for such purposes but not all devices support it, like the FC 43, unfortunately. Anyway from my modbus experience Usually register 0/1 are used for versioning the hw or the firmware of the device so I think that’s actually the best option. Maybe also requesting a coil could be a little bit faster but not every device support it.

---

Daniel