yaacov/node-modbus-serial #369: truncated messages on tcp port due to split data on socket event

rwrnet, 2020-10-12:
NodeJS: v14.13.1.

I chased an non-consistend issue of wrong data in a communication with an Inverter down to a clear reason with this simple example code:
```
var client = new net.Socket();
client.on('data', function(data) {
	console.log('Received: ' + data.toString("hex"));
});
client.connect(502, '10.0.2.244', function() {
	console.log('Connected');
	client.write(Buffer.from("000100000006030378530002","hex"));
});
```
The output is **sometimes**:
```
Received: 000100000007030304000013
Received: 88
```
So NodeJS Socket emits two messages but on Wireshark one can clearly see, this was sent by the inverter in one response:
```
0000   03 04 00 00 13 88
```
Unfortunately the tcpport receiver doesn't handle this correctly and instead creates a message of the original size (from the message header) which then get filled up by a 0-byte ending in a payload of `13 00` instead of the expected `13 88`:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L54

So I guess, the code needs to decouple Socket emits from Modbus emits. I'm neither a Modbus nor a nodejs crack, but if you don't mind to support me, I can work on a PR.

martinjamilis, 2021-12-07:
Hi,
I'm facing the same issue, did you find a solution?

rwrnet, 2021-12-09:
As a workaround, pull a byte more than needed and just ignore the extra byte when parsing the result. But I hate workarounds. ;)

AuspeXeu, 2023-01-27:
See #396