yaacov/node-modbus-serial #374: readRegisters catch an errorData length error?

g-zajac, 2020-11-07:
Hi
I got this error occasionally:

`
error: modbus.js readRegisters catch an errorData length error, expected 107 got 230 {"stack":"Error: Data length error, expected 107 got 230\n    at Modbu
sRTU._onReceive (../node_modules/modbus-serial/index.j
s:362:14)\n    at TcpPort.emit (events.js:315:20)\n    at Socket.<anonymous> (../node_modules/modbus-serial/ports/tcpport.js:77:20)\n    at Socket.emit (events.js:315:20)\n    at addChunk (_stream_readable.js:
295:12)\n    at readableAddChunk (_stream_readable.js:271:9)\n    at Socket.Readable.push (_stream_readable.js:212:10)\n    at TCP.onStreamRead (internal/s
tream_base_commons.js:186:23)"}
Data length error, expected 107 got 230
`

I need to reset device (Moons stepper motor) and nodejs server after that error to make the modbus working again.
Any idea how to prevent, fix it please?

yaacov, 2020-11-08:
Hi, thank you for the issue :+1:

the offending line is:
https://github.com/yaacov/node-modbus-serial/blob/b0a3cfd1bfc00a76bd8a06c516b1cee107421a02/index.js#L376

a- you can disable this check to get you code working.
b- I labelled this issue with help wanted in case someone knows the root cause of the problem and can make a better fix.

yaacov, 2020-11-08:
p.s.
If you find a fix ( better then disabling the check :-)  ), please comment here to help the next one hitting this issue.

g-zajac, 2020-11-08:
Hi, thank you for the quick answer. I have commented the check and than receive an other error:

```
error: modbus.js readRegisters catch an errorUnexpected data error, expected 1 got 1 {"stack":"Error: Unexpected data error, expected 1 got 1\n    at Modbu
sRTU._onReceive (../node_modules/modbus-serial/index.j
s:374:14)\n    at TcpPort.emit (events.js:315:20)\n    at Socket.<anonymous> (../node_modules/modbus-serial/ports/tcpport.js:77:20)\n    at Socket.emit (events.js:315:20)\n    at addChunk (_stream_readable.js:
295:12)\n    at readableAddChunk (_stream_readable.js:271:9)\n    at Socket.Readable.push (_stream_readable.js:212:10)\n    at TCP.onStreamRead (internal/s
tream_base_commons.js:186:23)"}
Unexpected data error, expected 1 got 1
```

no clue so far, will try to investigate further

g-zajac, 2020-12-07:
Hi, after further investigation It seems that the error is related to sending two modbus commands at the same time - my guess. In my app the server is reading registers of each device 3 times per second. On top of that user can change motor position - sending modbus write command. In 'lucky' situation both commands can happen in exactly the same time. I have reduced the reading registers frequency to 1sec and have not got the error so far. I'm going to add in the code a flag/lock so when user sends a modbus command the server holds with enquiring modbus read. What is your opinion? Is it the right direction?

Jarwain, 2020-12-23:
I can confirm that I have encountered the same issue when trying to send multiple modbus commands simultaneously. I've implemented two different workarounds for two different applications; one workaround involves writing a basic queue around the client such that the next request is only sent after a response is received from the first request. 
Our other workaround (which is probably better for your usecase) is to have one connection dedicated for reading registers and a second connection dedicated to writing registers. 

I'm not sure how to fix this issue within the library itself (could try implementing the queue within the library per the first workaround, but this reduces end-user control unless its made an option. Not sure if this is a direction @yaacov wants to take this lib though).

yaacov, 2020-12-24:
> Not sure if this is a direction @yaacov wants to take this lib though).

I'm always happy to make things more helpful and usable :-)

can you make it in a new directory with an api that is adding on top of current api so it will not break current usage cases ?
( see how https://github.com/yaacov/node-modbus-serial/tree/master/servers and https://github.com/yaacov/node-modbus-serial/tree/master/apis adds functionality without changing original behaviour )