yaacov/node-modbus-serial #375: Modbus RTU/TCP slave

hunter7793, 2020-11-09:
Hi.
can i using this driver and create a modbus(on RTU and TCP protocols) slave device?
if i can - tell me please how, if i cant - will you add this function?

Thank you.

esdrasjnr, 2020-11-09:
I'm trying to create a TCP slave device with this code [here](https://github.com/yaacov/node-modbus-serial/wiki/Example:--server#modbustcp-server). But when I request the slave value from my master, it cannot read the data and throws an error. In addition, someone has any idea what it could be?

PS. My master is another process, written in C

Jarwain, 2020-12-23:
A modbus slave is the same thing as a modbus server, the example @esdrasjnr links or [this one](https://github.com/yaacov/node-modbus-serial#modbustcp-server) in the docs should accurately represent how to use it. 

@esdrasjnr this isn't my repo but generally it's a good idea to open a separate ticket with your problem, as well as sample code. I've successfully set up a modbus TCP server using this library and have retrieved data from it using this library and a number of other modbus tools, so the issue is likely a problem with your modbus master/client implementation, and difficult to help debug without some sample code

1038642892, 2022-11-16:
I have a pull request in for an RTU server:
https://github.com/yaacov/node-modbus-serial/pull/480