yaacov/node-modbus-serial #376: How to get data from COM PORT

TheRelow, 2020-11-10:
Hi,
I have not found an example of using data from COM PORT. I tried to do something according to the examples found here, but I get "Unknown error code 3" or "Error: Opening COM1: Access denied"

`
        let serialPort = new serialport(item.pnpId, {baudRate: 9600});
        console.log(serialPort)

        client.open()
            .then((v)=>{
              console.log(v)
            })
            .catch((v)=>{
              console.log(v)
            })
`
For example here comes the error "Unknown error code 3"

yaacov, 2020-11-10:
Hi, thanks for the issue, can you show a more complete code that create the error ?

TheRelow, 2020-11-10:
> Hi, thanks for the issue, can you show a more complete code that create the error ?

Ofc

yaacov, 2020-11-10:
wow, didn't know it works from a browser, only tested with nodejs.

Try using the com name (e.g. "COM1") instead of the actual com object (and make sure it's close before you try to open it for comunication):
`client.connectRTU("COM1", {baudRate: 9600}).then ... // make sure com1 is not open by other user/method `

TheRelow, 2020-11-11:
> wow, didn't know it works from a browser, only tested with nodejs.
>
> Try using the com name (e.g. "COM1") instead of the actual com object (and make sure it's close before you try to open it for [...]

I have tried this connection. But I always get the error "Error: Opening COM4: Access denied"

yaacov, 2020-11-11:
yes, AFAIU browsers will not let access peripherals without permissions (e.g. camera, disk, serialPort ... )

You will need some way to get access, use some browser extension or library to access the serial port, see:
https://developer.chrome.com/apps/serial
Or build a service that run in the background, see:
https://github.com/yaacov/node-modbus-ws

Xsandor, 2022-01-13:
@yaacov, I believe your last comment explained both why this is happening and also how TheRelow should move forward. Close issue due to inactivity?

closed 2022-01-13 by yaacov