yaacov/node-modbus-serial #378: How write a 32 bit value (e.g. a float)?

matteodisabatino, 2020-12-16:
Hi,
Reading the code I've see that all values are converter to UInt16BE ([this is function FC6](https://github.com/yaacov/node-modbus-serial/blob/master/index.js#L736), in particular I'm referring to lines 766-770). So... what if I have to write a 32bit value such as a float?

Thanks in advance

yaacov, 2020-12-16:
Hi, thank you for the issue.

does this example help you ?
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js#L28

matteodisabatino, 2020-12-16:
Hi, thanks for the answer. I've read the example and I think is useful, but my problem is on writing, not on reading.

yaacov, 2020-12-16:
you need to do the same thing ... convert the float into buffer and send it using  `writeRegisters` that takes a buffer:

https://github.com/yaacov/node-modbus-serial/blob/master/ModbusRTU.d.ts#L48

matteodisabatino, 2020-12-16:
Yeah!! It worked 😍 
Thank you very much 😄

closed 2020-12-16 by yaacov

fulminetech, 2021-04-14:
Could you please share an example to write 32bit register? Thanks!

matteodisabatino, 2021-04-14:
Sure:

```js
client.writeRegisters(THE_REGISTER, THE_32_BIT_VALUE, (err, data) => {
    if (err) {
      throw err
    }

    console.log('OK: ', data)
  })
```

Just let THE_32_BIT_VALUE to be a buffer of a 32 bit value :-)