yaacov/node-modbus-serial #379: Unhandled 'error' event when disconnecting RS485 while reading registers

cakemasher, 2020-12-19:
Currently I'm using a Raspberry Pi together with a RS485 USB adapter. Everything is working fine, untill I disconnect the USB adapter while the Rspberry Pi is reading registers. The Serialport module throws the error 'Port is not open', but it is not caught by the Modbus-Serial module.

I'm not really familiar with GitHub syntax, and I'm not sure how to link to a specific line in a file, but the issue is on line 73 of file ports/rtubufferedport.js. Here a SerialPort object is created and on line 76 of the same file a listner is created on the data event. However, no 'error' listners are attached to the SerialPort object causing my script to crash upon a SerialPort error.

It would be awesome to let Modbus-Serial throw an error when SerialPort does.

I'm using serialport version 9.0.4 and modbus-serial version 8.0.1

yaacov, 2020-12-20:
Hi, Thank you for the issue :+1:

Can you make a pull request with a fix ?

yaacov, 2020-12-20:
fixed by #380

closed 2020-12-20 by yaacov