yaacov/node-modbus-serial #381: How to read/write first or second bit of a registrer

yususs, 2020-12-20:
Hi,
I need an example to operate on different bits of a one single register.
E.g. 
How to write `1` to first/second/third bit of register 2000 

Thanks

Jarwain, 2020-12-23:
Knowing that a register consists of 16 bits, we can represent these bits in a number of ways. `0xFFFF` for hex, `0b1111111111111111` in binary. 

If you want to change a single bit without changing any of the other bits, first you'll have to read the register using the `readInputRegisters` or `readHoldingRegisters` functions. You should receive the value of the register. You'll want to read that, either using the data property (i don't 100% remember what it returns, I think its the value represented as an integer/base10) or the buffer proprety (which gives you a nodeJS buffer). You'll want to write a function that can take either input, transform it to a more malleable format, figure out how to modify it to flip a single bit to that number (I recommend reading up on the [bitwise operators](https://stackoverflow.com/questions/1436438/how-do-you-set-clear-and-toggle-a-single-bit-in-javascript), and then write it back to the register using the `writeRegisters` function. 

Hope this is enough info for you to figure out what to google. Good luck!

yususs, 2021-01-01:
It help a lot. Thank you so much

closed 2021-01-01 by yususs