yaacov/node-modbus-serial #382: Modbus/TCP: If the connection is broken, we get an unhandled promise rejection that we can't catch

ryangriggs, 2020-12-22:
If I disconnect the modbus/tcp device from the network, to test what happens, I am getting the following error:

  (node:25171) UnhandledPromiseRejectionWarning: Error: TCP Connection Timed Out
      at Socket.<anonymous> (/home/pi/vision/node_modules/modbus-serial/ports/tcpport.js:111:24)
      at Socket.emit (events.js:315:20)
      at Socket._onTimeout (net.js:483:8)
      at listOnTimeout (internal/timers.js:554:17)
      at processTimers (internal/timers.js:497:7)
  (node:25171) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2041679)

This error seems to be un-catchable within my code, no matter where I add a try/catch block or promise .catch() statement, this error is still thrown.

yaacov, 2020-12-22:
hello, thank you for the issue !

can you make a pull request fixing that ?

ryangriggs, 2020-12-22:
Working on it...

ryangriggs, 2020-12-22:
From what I can tell, it appears this is a bug in request-promise-native (see https://github.com/request/request-promise-native/issues/31)

romor, 2022-06-26:
I face the same issue. Would be great if this could be resolved.