yaacov/node-modbus-serial #384: TypeError: SerialPort is not a constructor

ArcherGu, 2021-01-04:
Hi, I'm using electron environment and it showed this error when I started
`TypeError: SerialPort is not a constructor
    at ModbusRTU.cl.connectRTUBuffered (webpack:///./node_modules/modbus-serial/apis/connection.js?:250:22)`

yaacov, 2021-01-04:
hi, thank you for the issue,

I aded a "help wanted" label in case someone knows how to solve it,
If you find how to solve it, please comment here or open a pull request with the fix.

ArcherGu, 2021-01-04:
I seem to have solved my own problem.
1. `npm install serialport -S`
2. `electron-rebuild -f -w serialport` (add this to your package.json and run it)

I actually know I need electron-rebuild it, but I don't know why I need to install serialport extra.
Isn't serialport automatically installed? Is the version of serialport required by modebus-serial different from the current version?

yaacov, 2021-01-04:
> Isn't serialport automatically installed? 

it is:
https://github.com/yaacov/node-modbus-serial/blob/master/package.json#L41

> Is the version of serialport required by modebus-serial different from the current version?

it's `"serialport": "^9.0.0"`:
https://github.com/yaacov/node-modbus-serial/blob/master/package.json#L41

yaacov, 2021-01-04:
p.s. in npm you can also find a release without pre-packed serialport: https://www.npmjs.com/package/modbus-serial/v/8.0.1-no-serial-port

ArcherGu, 2021-01-04:
> p.s. in npm you can also find a release without pre-packed serialport: https://www.npmjs.com/package/modbus-serial/v/8.0.1-no-serial-port

Thank you very much for the answer.

closed 2021-01-07 by ArcherGu

mllw, 2021-11-22:
if use vue-cli-plugin-electron-builder
1、npm i serialport
2、 edit vue.config.js
add  externals: ['serialport']

module.exports = {
  pluginOptions: {
    electronBuilder: {
      externals: ['serialport']
    }
  }
}