yaacov/node-modbus-serial #385: Range error "ERR_OUT_OF_RANGE"

TrentiNic, 2021-01-07:
**Background**
I've implemented a connection on a tcp line using a ModbusRTU client:
`var client = new ModbusRTU();
client.connectTCP("127.0.0.1", { port: 8502 });`
and I'm reading data every 2 seconds:
`client.readHoldingRegisters(0, 10, function(err, data) {console.log(data.data); });`

**Description**
What I'm experiencing is an error "ERR_OUT_OF_RANGE" that shows after the application has run for a while (a tens of minutes).

**Example**
`RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of range. It must be >= 0 and <= 255. Received 256
    at writeU_Int8 (internal/buffer.js:729:11)
    at Buffer.writeUInt8 (internal/buffer.js:739:10)
    at ModbusRTU.writeFC4 (C:\nodeJS\node_modules\modbus-serial\index.js:568:9)
    at ModbusRTU.writeFC3 (C:\nodeJS\node_modules\modbus-serial\index.js:530:10)
    at ModbusRTU.converted [as readHoldingRegisters] (C:\nodeJS\node_modules\modbus-serial\apis\promise.js:36:27)
    at C:\nodeJS\main.js:713:20
    at Array.forEach (<anonymous>)
    at C:\nodeJS\main.js:705:15
    at TcpPort.callback (C:\nodeJS\node_modules\modbus-serial\index.js:407:17)
    at handleCallback (C:\nodeJS\node_modules\modbus-serial\ports\tcpport.js:46:20) {
  code: 'ERR_OUT_OF_RANGE'
}`

yaacov, 2021-01-07:
Hi, thank you for the issue.

If you find a solution, please write a pull request fixing it.

TrentiNic, 2021-01-07:
Found the issue! I was incrementing the ID of the client every time a new connection was open:
`client.setID("ID");`
I removed the auto increment and everything works fine now.

closed 2021-01-07 by TrentiNic