yaacov/node-modbus-serial #387: large holding register read possible Bug involving Data length error,

14drew, 2021-01-23:
Think there might be a some sort of byte roll over bug in the reading multiple holding registers response size checking as I get 
Data length error, expected 261 got 5
when from a log read (client.readHoldingRegisters(100, 128) in this case.
(Sorry if I presenting this wrong but new to this but wanted to contribute if I can as I appreciate being able to use open source code)
anyway 261-5 is 256 which makes me think the 5 is actually 261 but clipped to the lower 8 bits.

Now I'm just reading in smaller blocks to avoid this issue but wanted to point this out anyway

yaacov, 2021-01-23:
Thank you for the issue.

If you find a fix please comment here with some notes, or post a pull request.

sigma-design, 2021-01-23:
Please note that according to Modbus Specifications (V1.1b3) the valid range for reading Holding Registers is 1...125, so the attempt to read 128 registers is invalid by definition.

14drew, 2021-01-24:
OK sorry for my mistake, I see fn3 response is limited to an 8 bit byte length now.
I guess the simple modbus TCP client (and the server) I was using to test must be using the 16 bit TCP packet byte length to buck the lower level modbus specification as that read the registers OK.
Thanks for the pointer.

closed 2021-01-24 by 14drew

yaacov, 2021-01-24:
@14drew it's not a mistake to ask :green_heart:  , if you want to play, you can try to catch this error in the appropriate place and make a pull request to return a more helpful error message before we get to the data length error, so the next person hitting this issue will have better error message.