yaacov/node-modbus-serial #389: Security vulnerabilities found for the serialport library being used

Sebanti13, 2021-02-10:
The following security vulnerabilities were found for the serialport library used by modbus-serial:
1. A vulnerability was found in the **serialport_win.cpp** file of serialport module.
The dangerous function, strncpy, was found in use at multiple lines 58 in the @serialport/bindings/src/serialport_win.cpp file. Such functions may expose information and allow an attacker to get full control over the host machine.
The function strncpy_s is a recommended security replacement for the strncpy function.

2. A vulnerability was found in the **nan.h** file of nan module under serialport.
The dangerous function, memcpy, was found in use at one place in nan/nan.h file. Such functions may expose information and allow an attacker to get full control over the host machine.
The function memcpy_s is a recommended security replacement for the memcpy function.

yaacov, 2021-02-10:
Thanks :+1:
can you make a pull request bumping the serial port version to a safe one ?

Sebanti13, 2021-02-11:
Actually we are not completely sure of which serialport version contains a fix, if it does, this was identified while running the Cybersecurity tests by our team. I have raised it here because modbus-serial is the direct dependency we are using and serialport was a sub dependency.
I will inform the same to the serialport team. Could you incorporate that version of serialport when the team releases?

yaacov, 2021-02-11:
hi
we have an npm package that does not use serial port, so if you don't actually need the serial port functionality you can use this:
https://www.npmjs.com/package/modbus-serial/v/8.0.1-no-serial-port

 > I will inform the same to the serialport team. Could you incorporate that version of serialport when the team releases?
 
 sure, I will be happy to update

Sebanti13, 2021-02-15:
Hi, Thank you for the suggestion. We have looked into it. Actually we will require serialport for communication with the serial device. 
I have communicated the same issue to the serialport library. :)