yaacov/node-modbus-serial #390: Modbus RTU sniffer

tenisak, 2021-02-16:
I would need to use node.js to listen for Modbus RTU communication between two other devices. Is this possible? I don't know if the modbus-serial library can be used for this.
My idea is to open the serial port and route the data from rx line directly to the parser. I will not send requests, I will only get responses. How to do it? :-)

yaacov, 2021-02-16:
Hi, thanks for the question

You should use some serial port sniffing tool, that expose a virtual port that sniffs the port/s you are using for communication 
this is a random link I found searching for serial sniffer: https://www.cyberciti.biz/faq/howto-monitor-data-on-a-serial-port-in-linux/

tenisak, 2021-02-16:
Thanks a lot for the quick response! Opening a serial port and getting raw data is easy... Rather, I'm looking for a way to use the parser you have inside modbus-serial. Is it possible to use [_onReceive function](https://github.com/yaacov/node-modbus-serial/blob/d5b60c979ee1a890aa53f25615c90acfc428b33c/index.js#L291) somehow? Insert any of my functions into transaction.next and continue by [CRC check](https://github.com/yaacov/node-modbus-serial/blob/d5b60c979ee1a890aa53f25615c90acfc428b33c/index.js#L354), etc.? :-)

yaacov, 2021-02-16:
No, but you can create one :-)
you can create a port that does that, for the client side look at how works https://github.com/yaacov/node-modbus-serial/blob/master/ports/rtubufferedport.js 
it's not exactly what you are looking for but it may be a starting point.

tenisak, 2021-02-16:
Thanks so much for the starting point. I will try to deal with it :-)