yaacov/node-modbus-serial #393: Modbus ASCII not receiving reply from slaves

ThKls, 2021-02-22:
I couldn't manage to retrieve data from Modbus slave using Modbus ASCII. E.g. readHoldingRegisters generates a timeout.

Looking into the code I found a bug in asciiport.js, line 133:
`var sdIndex = modbus._buffer.indexOf(0x3A); // ascii for ':'`
This can't be correct. The standard stated the reply from the slave starting with '>' (0x3E) instead of ':' (0x3A), which is for the request.
I changed it to:
`var sdIndex = modbus._buffer.indexOf(0x3E); // ascii for '>', Fix by ThKls`
Now I got correct replies.
Don't know if this is necessary for building Modbus slaves.

yaacov, 2021-02-22:
Hi, thank you for the issue :green_heart: 

Can you make a pull request with a fix ?

https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request
https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/about-forks
https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

ThKls, 2021-02-22:
Done

ThKls, 2021-03-01:
Closed with #395

closed 2021-03-01 by ThKls