yaacov/node-modbus-serial #396: What is the diff between TcpPort and TcpRTUBufferedPort

yarosdev, 2021-03-05:
Could you explain use cases ?

yaacov, 2021-03-05:
Thanks for the question :+1: 

TcpPort, is naive, it expects the complete modbus frames to come in one TCP packet, this is usually the case in small networks.
TcpRTUBufferedPort can buffer data coming over TCP channel and build complete modbus frames from several disconnected TCP , this is useful in bigger network that split the message over several routes.

use cases:
TcpPort - use on small networks
TcpRTUBufferedPort - use on large networks, or if TcpPort has lots of errors related to receiving broken modbus frames (e.g. CRC erros).

yarosdev, 2021-03-06:
It seems that TelnetPort is also buffered. Why dont you called it TelnetRTUBufferedPort ?

AuspeXeu, 2023-01-27:
Why not have only a buffered version, what is the advantage of TcpPort over its buffered counter part?