yaacov/node-modbus-serial #400: how to read registers from meter?

dyaacov, 2021-03-22:
Hi Guys,

I have a meter and a modbus server (I used some old npm library).

The server is listening to incoming connections from the meter and upon an incoming connection, I call connection.readInputRegisters..

How can I implement the same with this library?
In the examples, the "server" is in charge to provide the data while in my case I need to pull data from the meter.

Thanks
Dekel

yaacov, 2021-03-22:
Hi, thanks for the question :+1: 

Please take a look at:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/simple.js

dyaacov, 2021-03-22:
Thanks Yaacov,

I'm not sure how it helps.

in the example you sent, the client open connection and request for registers.
in my case, the meter opens connection to the server and the server needs to ask for registers from the meter on the open socket.

Thanks
Dekel

yaacov, 2021-03-22:
Hi, hmm

So you have an open socket and you want to use it for communication ?

you can try and use `linkTcpRTUBuffered`:
https://github.com/yaacov/node-modbus-serial/blob/master/apis/connection.js#L174