yaacov/node-modbus-serial #403: How do I set the slave ID

ailen1002, 2021-04-07:
15/5000 
After establishing a connection, I want to read data from slave station 2。

yaacov, 2021-04-07:
Thank you for your question, you can look at the examples directory, for example:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/simple.js#L27

ailen1002, 2021-04-08:
I've already set it，My code:
export function connectTCP() {
  // client.close();
  client.setID(2);
  console.log('AAAAAAAAAAAAAAAAAAAAAAA: ' + client.getID());

  client.setTimeout(mbsTimeout);
  client.connectTCP(mbsHost, { port: mbsPort})
    .then(() => {
      mbsState = MBS_STATE_GOOD_CONNECT
      mbsStatus = "Connected, wait for reading...";
      console.log(mbsStatus);
    })
    .catch((e: { message: string; }) => {
      mbsState  = MBS_STATE_FAIL_CONNECT;
      mbsStatus = e.message;
      console.log(e);
    })
}

export function readModbusData() {
  client.readHoldingRegisters (0, 10)
    .then((data) => {
      mbsState   = MBS_STATE_GOOD_READ;
      mbsStatus  = "success";
      console.log(data);
      return data
    })
    .catch((e) => {
      mbsState  = MBS_STATE_FAIL_READ;
      mbsStatus = e.message;
      console.log(e);
    });
}
but  Always read Mbslave1
Here is my command line print message:
ping
AAAAAAAAAAAAAAAAAAAAAAA: 2
Connected, wait for reading...
pong
{
  data: [
    0, 10,  3, 0, 40,
    0,  0, 20, 4,  5
  ],
  buffer: <Buffer 00 00 00 0a 00 03 00 00 00 28 00 00 00 00 00 14 00 04 00 05>
}
my Mbslave2 date is [1,2,3,4,5,6,7,8,9,0]

yaacov, 2021-04-08:
try to set the client ID after connection is ready, see the examples:
```jsclient.connectTCP("127.0.0.1", { port: 8502 })
    .then( // set client id )
    .then( // get data from client
```