yaacov/node-modbus-serial #404: Error with two simultan request for Modbus tcp

HaKePlan, 2021-04-16:
I found something very interesting, where I ask myself, why this works in this way...

When we have two requests for Modbus TCP at the same time, it cast's an error.
The code where I found this problem is simple:

There is, let's call it the Modbus handler, and the API server, which sends the result from the Modbus handler to the client (very simplified)...

The Modbus handler does for each request on the API server the same thing:

```
// there is the initial stuff:

// create an empty modbus client
const client = new ModbusRTU();

// set the defined offset fpr the address translation
dotenv.config({ path: './config.env' });
const offset = process.env.MODBUS_OFFSET * 1;

// conect function
const connectClient = async () => {
  await client.connectTCP(process.env.MODBUS_IP, {
    port: process.env.MODBUS_PORT,
  });

  // console.log('modbus connection enabled');
  client.setID(1);
  client.setTimeout(1000);
};

// then, there is the code that gets called for each request on the API server

// first open connection
await connectClient();

// get the address and register from req
const address = req.params.address

// get value from Modbus TCP
const val = await client.readCoils(address + offset, 1);

// close connection
client.close();

// return value
return val
```

The API server is an express API server
The client is a next.js application that gets data in a 5-second interval for several Modbus entries from the API server.

Exactly there happens the error.
The next.js application sends several requests for different Modbus entries at the same time. The API server calls the Modbus handler and awaits the value to send it back to the app. But instead of the value, the `client.readCoils(address + offset, 1);` function returns an error like this:

`Error: Port Not Open`

I don't understand why this is happening. As far as I know, node.js should handle each request in a separate node. So, the `client.close();` shouldn't affect other, still running ,`client.readCoils`? But also, as far as I understand the error, `client.close();` interrupts the other running `client.readCoils`?

I am really looking for an answer to solve this issue. It is necessary for the app to make several request at the same time.

yaacov, 2021-04-17:
Hi, thank you for the issue, I added a help wanted flag in case someone knows the answer and can help

HaKePlan, 2021-04-17:
@yaacov 

I think I found a possible solution...
Is there any problem, when I call the `client.connectTCP()` at the start (on npm start) of the API server?
This means the connection to the Modbus client will be open as long as the API server runs.

yaacov, 2021-04-18:
You can also try to use a socket you create:
https://github.com/yaacov/node-modbus-serial/blob/master/apis/connection.js#L121

HaKePlan, 2021-04-19:
Opening the connection at the start of the app with client.connectTCP() works perfectly fine. I didn't get any errors at all. I think I can close the issue.

closed 2021-04-19 by HaKePlan