yaacov/node-modbus-serial #405: TransactionTimedOutError cannot be caught when using async/await

Jalle19, 2021-04-19:
Not sure why but it's impossible to catch `TransactionTimedOutError` errors, they always end up triggering an unhandled rejection warning no matter where you place catch clauses.

After hours of debugging I resorted to giving up.

yaacov, 2021-04-19:
HI, thanks for the issue :green_heart: 

Never give up :-)
Can you paste-here/link-to-gist a small demo program that simulates the missing error ?
It may help other people looking at this issue to have an idea how to help ?

Jalle19, 2021-04-19:
It's hard to reproduce, I think the easiest way is to set a really low timeout. Currently I'm using 5 seconds, and the error happens only once a day or something like that.

https://github.com/Jalle19/eda-modbus-bridge/blob/master/app/modbus.mjs#L29 here's an example of a call that (I think) triggers the error. Any of the interactions with `modbusClient` can seemingly trigger it.

I'm not entirely sure if the `async-mutex` library I'm using to guard access to the serial port is the real culprit here, but it's certainly possible?

Jalle19, 2021-04-19:
The most I've been able to do when it comes to logging the error is this:

```
Apr 18 09:56:53 homeassistant b5a380b3f13b[390]: (node:204) UnhandledPromiseRejectionWarning: #<TransactionTimedOutError>
Apr 18 09:56:53 homeassistant b5a380b3f13b[390]:     at emitUnhandledRejectionWarning (internal/process/promises.js:170:15)
Apr 18 09:56:53 homeassistant b5a380b3f13b[390]: (node:204) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 22)
Apr 18 09:56:53 homeassistant b5a380b3f13b[390]: (node:204) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code
```

pfried, 2021-11-03:
Had the same issue but imho it has nothing to do with the library but with the nature of `async / await` as you can read here: https://itnext.io/error-handling-with-async-await-in-js-26c3f20bc06a

Jalle19, 2023-08-14:
Turns out this was PEBKAC :shrug: