yaacov/node-modbus-serial #406: Global Variable

lekhoi99, 2021-04-22:
How can I set Data from Client.readHoldingRegisters as Global Variable ?

yaacov, 2021-04-22:
Hi, thank you for the question, I setup the "help wanted" label in case someone can help.

yaacov, 2021-04-22:
can you post a little example code, explaining what you want to do ?

lekhoi99, 2021-04-22:
I want to get Data from Modbus RTU and then put it to OPC UA Protocol
My code :
const getMeterValue = async (id) => {
    try {
        // set ID of slave
        await client.setID(id);
        // read the 1 registers starting at address 0 (first register)
        let val = await client.readHoldingRegisters(23, 37);
        // return the value
    } catch (e) {
        // if error return -1
        return -1
    }
}

const sleep = (ms) => new Promise(resolve => setTimeout(resolve, ms));

// start get value
getMetersValue(metersIdList);

// node OPC UA
const { OPCUAServer, Variant, DataType, StatusCodes } = require("node-opcua");
(async () => {

    // Let's create an instance of OPCUAServer
    const server = new OPCUAServer({
        port: 4880, // the port of the listening socket of the server
        buildInfo: {
            productName: "MySampleServer1",
            buildNumber: "7658",
            buildDate: new Date(2014, 5, 2)
        }
    });
    await server.initialize();
    console.log("initialized");

    const addressSpace = server.engine.addressSpace;
    const namespace = addressSpace.getOwnNamespace();

    // declare a new object
    const device1 = namespace.addObject({
        organizedBy: addressSpace.rootFolder.objects,
        browseName: "Inverter 1"
    });

    namespace.addVariable({
        componentOf: device1,
        browseName: "Frequency Output",
        dataType: "Double",
        value: {
            get: () => new Variant({ dataType: DataType.Double, value: val.data[0] })
        }
    });

    namespace.addVariable({
        componentOf: device1,
        browseName: "Speed",
        dataType: "Double",
        value: {
            get: () => new Variant({ dataType: DataType.Double, value: val.data[1] })
        }
    });

    namespace.addVariable({
        componentOf: device1,
        browseName: "Current",
        dataType: "Double",
        value: {
            get: () => new Variant({ dataType: DataType.Double, value: val.data[2] })
        }
    });

    const device2 = namespace.addObject({
        organizedBy: addressSpace.rootFolder.objects,
        browseName: "Inverter 2"
    });

    server.start(function () {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);
        const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
        console.log(" the primary server endpoint url is ", endpointUrl);
    });

})();

But val isn't Defined

> can you post a little example code, explaining what you want to do ?

yaacov, 2021-04-22:
did you try defining val as a global var ?
``` js
let var = ...
...
   var = await client.readHoldingRegisters(23, 37);
...
```

lekhoi99, 2021-04-22:
> did you try defining val as a global var ?
> 
> ```js
> let var = ...
> ...
>    var = await client.readHoldingRegisters(23, 37);
> ...
> ```

I tried. But it doesn't work

closed 2021-04-24 by lekhoi99