yaacov/node-modbus-serial #421: TransactionTimedOutError with USB2Serial Driver

Daschmi, 2021-07-14:
Hi,

i have a Solair 3100 (https://www.manualslib.com/manual/1732979/Lighthouse-Solair-3100.html?page=150) and wan't to communicate over usb with the device. The device uses the modbus protocol and i try the driver from the manufacturer.

I try to read register 40001 (2 Bytes?) 

```var ModbusRTU = require("modbus-serial");
var client = new ModbusRTU();

function run(error) {

    console.log("RUN", error);
    
    client.setID(1);
    client.setTimeout(2000);
     
    client._debugEnabled = true;
 
    client.writeFC3(1, 40001, 2, function(err, data) {

        console.log(err, data);

    });
 
}

/* <Buffer 01 03 9c 41 00 c8 3a 18> */

client.connectAsciiSerial(
    "COM7", 
    { 
        baudRate: 19200, dataBits: 8, stopBits: 1, parity: 'none'
    },
    run
);
```

I could open the connection, but i don't recive any bytes. Output with debug:

```
RUN null
TransactionTimedOutError {
  name: 'TransactionTimedOutError',
  message: 'Timed out',
  errno: 'ETIMEDOUT',
  modbusRequest: <Buffer 01 03 9c 41 00 02 ba 4f>,
  modbusResponses: []
} undefined
```

Does anyone have any idea what I am doing wrong?

yaacov, 2021-07-14:
note: we have another issue about ascii port:
https://github.com/yaacov/node-modbus-serial/pull/236/files
may be related.

Daschmi, 2021-07-14:
Thanks for your quick reply, but unfortunately the _asciiDecodeResponseBuffer function is not called at all.