yaacov/node-modbus-serial #427: can't use with serialport

vae1970, 2021-09-27:
if i use  modbus-serial and modbus-serial in electron, i will get result in electron-rebuild like this :
`binding.gyp not found while trying to load binding.gyp`
i also read issue in modbus-serial, i know modbus-serial used serialport, so how to use serialport's api , or how to use modbus-serial and modbus-serial together in electron

yaacov, 2021-09-29:
Hi, thank you for the question.
I added a "help wanted" label to let people know the can help answer, you can also try asking electron developers if they know of an issue and may help.

yaacov, 2021-12-09:
did you try the package that does not import serialport implicitly ?
https://www.npmjs.com/package/modbus-serial/v/8.0.5-no-serial-port

mllw, 2021-12-20:
in package.json
add this script

"rebuild": "electron-rebuild -v 13.0.0 -f -w serialport",

work good for me

Xsandor, 2022-01-13:
I can confirm that the fix suggested by @mllw solves the problem.

I just had to change 13.0.0 to 16.0.0 to match the Electron version that I'm using.

yaacov, 2022-01-13:
Thanks all :green_heart:

closed 2022-01-13 by yaacov

Kocinkag, 2022-02-22:
Merhaba @yaacov
"modbus-serial" kütüphaneni çok sevdim ama "electronJS" ile kullandığımda bir çok hata aldım. Sorunu uzun uğraşlar sonucu buldum. "modbus-serial" kütüphanesinde "serialport" kütüphanesini son sürüme güncellersen ve "serialport" kütüphanesine bağlı sınıfları güncellersen sorun düzelecektir :)

Hi @yaacov
I loved your "modbus-serial" library, but when I used it with "electronJS" I got a lot of errors. I found the problem after a long struggle. If you update the "serialport" library to the latest version in the "modbus-serial" library and update the classes that depend on the "serialport" library, the problem will be fixed :)