yaacov/node-modbus-serial #432: Modbus ASCII not receiving response

OlivierVerhaegen, 2021-11-05:
Hi,

We where testing Modbus ASCII communication with a Lighthouse Remote 5104 particle counter.
We get a Timeout error because no response is received.
[image]

This is our code:
```javascript
const Modbus = require("modbus-serial");

const client = new Modbus();
client.setTimeout(2000);
client.setID(1);

client.connectAsciiSerial('COM4', {
    baudRate: 19200,
    dataBits: 8,
    stopBits: 1,
    parity: "none",
}).then(() => {

    console.log('connected');

    client.readHoldingRegisters(27, 1).then(console.log).catch(console.error);
});
```

We've directly used the [serialport](https://github.com/serialport/node-serialport) library to check if communication works that way and it does 👍 . 
This is our code:

```javascript
const serial = require('serialport');

const buffer = new Buffer.alloc(17);
buffer[0] = 0x3A;

buffer[1] = 0x30;
buffer[2] = 0X31;

buffer[3] = 0x30;
buffer[4] = 0x33;

buffer[5] = 0x30;
buffer[6] = 0x30;
buffer[7] = 0x31;
buffer[8] = 0x42;

buffer[9] = 0x30;
buffer[10] = 0x30;
buffer[11] = 0x30;
buffer[12] = 0x31;

buffer[13] = 0x45;
buffer[14] = 0x30;

buffer[15] = 0x0D;
buffer[16] = 0x0A;

console.log(buffer.toString('ascii'));

const com = new serial('COM4', {
    baudRate: 19200,
    dataBits: 8,
    stopBits: 1,
    parity: 'none',
    autoOpen: false
});

com.open(function(err) {
    if (err) {
        console.error(err);
    } else {
        process.stdin.resume();
        console.log('Port open');
        setInterval(function() {
            com.write(buffer, function(err) {
                if (err) console.error(err);
            });
        }, 2000);
    }
});

com.on('readable', function() {
    console.log('Data: ', com.read());
});
```

We followed: https://www.simplymodbus.ca/ASCII.htm

We suspect something is going on in the `_asciiEncodeRequestBuffer` function in `ports/asciiport.js`.

Version: 8.0.4

yaacov, 2021-11-07:
hi, thank you for the issue :white_flower: 

This is the code for the ascii comunication:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/asciiport.js
If you can identify the problem please comment here 

we have two other issues with ASCII, does this issues relate to your issue ?
https://github.com/yaacov/node-modbus-serial/issues/235
https://github.com/yaacov/node-modbus-serial/issues/421

OlivierVerhaegen, 2021-11-08:
Hi Yaacov,

Thank you for your quick response!

I've investigated the `asciiport.js` code. 
Our response starts with the 0x3A (:) delimiter while a check is being done for a 0x3E (>) delimiter. 
https://www.simplymodbus.ca/ASCII.htm only mentions a colon (:) delimiter, could you share the reference you used to create this code?

Could you elaborate on the `modbus._length` property and `_checkData` function? I see that the `_length` property is being set to 0 in `AsciiPort` but it's never set after this. My concern is it being used in the `_checkData` function as expected length.

```javascript
var AsciiPort = function(path, options) {
    var modbus = this;

    // options
    options = options || {};

    // disable auto open, as we handle the open
    options.autoOpen = false;

    // internal buffer
    this._buffer = Buffer.from("");
    this._id = 0;
    this._cmd = 0;
    this._length = 0;

    // create the SerialPort
    this._client = new SerialPort(path, options);

    // register the port data event
    this._client.on("data", function(data) {

        // add new data to buffer
        modbus._buffer = Buffer.concat([modbus._buffer, data]);

        modbusSerialDebug({ action: "receive serial ascii port", data: data, buffer: modbus._buffer });
        modbusSerialDebug(JSON.stringify({ action: "receive serial ascii port strings", data: data, buffer: modbus._buffer }));

        // check buffer for start delimiter
        var sdIndex = modbus._buffer.indexOf(0x3E); // ascii for '>', as this indicates slave replies
        if(sdIndex === -1) {
            // if not there, reset the buffer and return
            modbus._buffer = Buffer.from("");
            return;
        }
        // if there is data before the start delimiter, remove it
        if(sdIndex > 0) {
            modbus._buffer = modbus._buffer.slice(sdIndex);
        }
        // do we have the complete message (i.e. are the end delimiters there)
        if(modbus._buffer.includes("\r\n", 1, "ascii") === true) {
            // check there is no excess data after end delimiters
            var edIndex = modbus._buffer.indexOf(0x0A); // ascii for '\n'
            if(edIndex !== modbus._buffer.length - 1) {
                // if there is, remove it
                modbus._buffer = modbus._buffer.slice(0, edIndex + 1);
            }

            // we have what looks like a complete ascii encoded response message, so decode
            var _data = _asciiDecodeResponseBuffer(modbus._buffer);
            modbusSerialDebug({ action: "got EOM", data: _data, buffer: modbus._buffer });
            if(_data !== null) {

                // check if this is the data we are waiting for
                if (_checkData(modbus, _data)) {
                    modbusSerialDebug({ action: "emit data serial ascii port", data: data, buffer: modbus._buffer });
                    modbusSerialDebug(JSON.stringify({ action: "emit data serial ascii port strings", data: data, buffer: modbus._buffer }));
                    // emit a data signal
                    modbus.emit("data", _data);
                }
            }
            // reset the buffer now its been used
            modbus._buffer = Buffer.from("");
        } else {
            // otherwise just wait for more data to arrive
        }
    });

    /**
     * Check if port is open.
     *
     * @returns {boolean}
     */
    Object.defineProperty(this, "isOpen", {
        enumerable: true,
        get: function() {
            return this._client.isOpen;
        }
    });

    EventEmitter.call(this);
};
```

```javascript
function _checkData(modbus, buf) {
    // check buffer size
    if (buf.length !== modbus._length && buf.length !== 5) {
        modbusSerialDebug({ action: "length error", recive: buf.length, expected: modbus._length });

        return false;
    }

    // check buffer unit-id and command
    return (buf[0] === modbus._id &&
        (0x7f & buf[1]) === modbus._cmd);
}
```

yaacov, 2021-11-08:
hi, 

I'm missing this line from your example:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/asciiport.js#L113

Is this change related to your issue ?

OlivierVerhaegen, 2021-11-08:
Hi Yaacov,

Yes this change is directly related to my issue.
Would it be possible to release a new version to npm?

yaacov, 2021-11-08:
https://www.npmjs.com/package/modbus-serial/v/8.0.5
:tada: :cake: :fireworks:

OlivierVerhaegen, 2021-11-08:
Thank you 🥳!! I'll verify the fix tomorrow.

OlivierVerhaegen, 2021-11-09:
Verified ASCII communication using the code below:

```javascript
const Modbus = require("modbus-serial");

const client = new Modbus();
client.setTimeout(2000);
client.setID(1);

client.connectAsciiSerial('COM4', {
    baudRate: 19200,
    dataBits: 8,
    stopBits: 1,
    parity: "none",
}).then(() => {

    console.log('connected');
    setInterval(function () {
        console.log('Reading...')
        client.readHoldingRegisters(27, 1).then(console.log).catch(console.error);
    }, 2000);
});
```

[image]

closed 2021-11-09 by OlivierVerhaegen