yaacov/node-modbus-serial #438: highbyte-lowbyte error

PJBCoetzer, 2021-12-17:
I have the situation where a supplier is packing the register data in a different byte order (lower byte first), so the value received from the `readHoldingRegisters()` call is not giving correct data. Would it be an option to add a optional parameter to specify the byte packing method (i.e. highbyte (default) or lowbyte) as a parameter in the framework ?

From https://www.simplymodbus.ca/FAQ.htm

**What is byte and word ordering?**
The Modbus specification doesn't define exactly how the data is stored in the registers. Therefore, some manufacturers implemented modbus in their equipment to store and transmit the higher byte first followed by the lower byte. (AE before 41).
Alternatively, others store and transmit the lower byte first (41 before AE).

yaacov, 2021-12-17:
Are you using the buffered data ?
https://github.com/yaacov/node-modbus-serial/blob/master/examples/polling_TCP.js#L74

You can parse the buffer anyway you need, in this example example the buffer is parsed as floats::
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffer.js#L30

PJBCoetzer, 2021-12-20:
Hi @yaacov,

Thanks, that makes sense. I was using the data array and totally missed the buffer output.
I will use the buffer to then implement the highbyte-lowbyte conversion.

Thanks for pointing this out.

closed 2021-12-20 by PJBCoetzer