yaacov/node-modbus-serial #440: ModbusRTU - 8th bit always false on writeCoils (FC15)

Rafael-Dagostim, 2022-01-10:
Hi, I've been using this library to comunicate via ModbusRTU to a STM32 but every time I use the function writeCoils, the 8th bit always return false on readCoils. The code I'm using:

```js
const ModbusRTU = require('modbus-serial');

const client = new ModbusRTU();

const delay = (ms) => new Promise((resolve) => setTimeout(resolve, ms));

async function read() {
  const result = await client.readCoils(0, 24);
  console.log(result);

  client.close(err => {
    if (err) console.log(err)
    else console.log('Desconectado');
  });
  await delay(3000);
}

async function write() {
  client.setTimeout(3000);
  client.setID(10);

  client.writeCoils(0, [
    true, true, true, true, true, true, true, true, true, true, // 10 bits
  ])

  await delay(3000);

  read();
}

client.connectRTUBuffered("COM3",
  {
    baudRate: 38400,
    dataBits: 8,
    stopBits: 1,
    parity: 'none',
  },
  write
)
```

I want know if it's normal happening or there's something that I can do to send everything by writeCoils without need to split in small parts (It's what I'm doing right now).

yaacov, 2022-01-10:
Hi, thank you for the issue,
Can you make a pull request fixing this behavior ?

Rafael-Dagostim, 2022-01-11:
I am not entirely familiar with the library but I will check if can solve this. Thanks.

Rafael-Dagostim, 2022-01-17:
After creating the modbus structure and sending it using directly the serialport library I noticed the same behavior so, I suppose this issue is not in this library. I still testing if it's not a problem on any physical device or in the serialport library. Anyway, as a workaround I'm using a for statement with writeCoil (FC5). Feel free to close this issue.