yaacov/node-modbus-serial #447: Modbus RTU over TCP

revo91, 2022-03-19:
Hi,
is it possible to use this library with Modbus RTU over TCP? I tried to do it using `connectTcpRTUBuffered ` but it wasn't working. I'm using [this converter](https://www.waveshare.com/rs485-to-eth.htm). The frame is from TCP rather than RTU.

mohammed-papad, 2022-04-13:
use connectTelnet
it's similar to Modbus RTU over TCP

zungzeny808, 2024-11-06:
client.connectTCP("127.0.0.1", { port: 512 });
 for modbus TCP protocol 
 
[image]

 client.connectTelnet("127.0.0.1", { port: 512 });
  for modbus OVERT TCP protocol 
  
example  modbus RTU over TCP package
[image]