yaacov/node-modbus-serial #461: Port not open error on TCP connection

Naim-Bijapure, 2022-06-07:
hi, I am getting a port not found error on the TCP connection.

actually, we are fetching plc data every 1 minute. the issue is coming after 6-8 hours. I am getting port not found error .
and after node server restart it starts again !

is there any solution for this?
i am on version 8.0.5

please help

yaacov, 2022-06-07:
hi, thank you for the issue :green_heart: 
I have added a "help wanted" in case someone knows how to fix it.

can you try and look for the offending code ?
can you try and fix it ?

Naim-Bijapure, 2022-06-07:
@yaacov  yes i can . but the issue is coming on real plc connection. it's perfectly working on the simulator. 
if you give me some pointers I can take a look.

yaacov, 2022-06-07:
sounds like the plc disconnects from time to time,
if this is the error, you may solve it by using a socket/port that can detect a "socket not found" error and tries to re-connect ?
if this is the problem, and if reconnecting solve it, a way forward can be to add an option to the TCP port to "reconnect on error"?
( I don't know the problem source :-) )

arnolddozsa, 2022-08-02:
Same issue here. 
The app automatically tries to reconnect, but gives Port not open error. 
When i shut down the app and restart it connects again without any issues. 
The problem occurs when a tcp connection looses or there is a tcp error and tries to reconnect. 
Did you find something workaround?