yaacov/node-modbus-serial #463: Modbus Serial communication is very unreliably working on my device

acroknacks, 2022-06-23:
It sometimes reads the data sometime, it hangs to read from the bus and write to the Bus.
I get a lot of CRC error while reading and writing to Register.
I am continuously reading from a coil, that also sometimes not read properly, and hangs.
I keep getting following error.
(node:4278) UnhandledPromiseRejectionWarning: Error: CRC error
    at RTUBufferedPort.<anonymous> (/home/acrodev29/Workspace/duvall/node_modules/modbus-serial/index.js:324:42)
    at emitOne (events.js:116:13)
    at RTUBufferedPort.emit (events.js:211:7)
    at RTUBufferedPort._emitData (/home/acrodev29/Workspace/duvall/node_modules/modbus-serial/ports/rtubufferedport.js:108:10)
    at SerialPort.onData (/home/acrodev29/Workspace/duvall/node_modules/modbus-serial/ports/rtubufferedport.js:71:22)
    at emitOne (events.js:116:13)
    at SerialPort.emit (events.js:211:7)
    at addChunk (_stream_readable.js:263:12)
    at readableAddChunk (_stream_readable.js:250:11)
    at SerialPort.Readable.push (_stream_readable.js:208:10)
(node:4278) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:4278) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

yaacov, 2022-06-23:
Hi, thank you for the issue
I added a need help label in case someone can help you

Can you add some information about the application?
Boud rate? Line type? Other devices on the line?
The code you use to connect and read? Timeouts you use?
Any other information you think can help someone that will try to help?

acroknacks, 2022-06-24:
I am using a velocio RTU Modbus Master, which has Registers and Coils mapped as required by Application.
Application runs on a Toradex arm64 device, with electron. Node version on device is 6.5.0
So communication is like, Application from Toradex , reads and writes Holding and Coil registers.
Application continuously keep reading one coil at the interval of 500 milli seconds. That read is not consistent. Sometimes it works and sometimes it does not.
Baud Rate - 9600, connected on Serial port.
console.log(config);

const client = new ModbusRTU();

// eslint-disable-next-line no-unused-vars
function serial_connection() {
  client.connectRTUBuffered(config.port, {
    baudRate: 9600,
    dataBits: 8,
    parity: "none",
    stopBits: 1
  });
  client.open();
  console.log('Connection Done');
  client.setID(1);
}

Timeouts not used.