yaacov/node-modbus-serial #467: writeCoil doesn't work while not giving any error

bkorumaz, 2022-07-03:
This code block doesn't work while it is not giving any error. I am able to write to this coil with the same IP adress, port and coil adress on QModMaster Application. What would be causing not to write to this coil?

I have put time interval between steps related to this topic #457 

[image]

yaacov, 2022-07-04:
HI, thank you for the question,
if you want to catch an error, you need to use `.catch`
see: https://github.com/yaacov/node-modbus-serial/blob/master/examples/write.js#L27

bkorumaz, 2022-07-04:
I successfully write to coil when I changed the adress with 2049. I should have write 2049 to address in order to change the value stored in the coil addressed 2050.

closed 2023-02-23 by bkorumaz