yaacov/node-modbus-serial #482: fail om tsc buid with version 8.0.7

dbettoni, 2022-11-25:
On last version 8.0.7 my tsc build produce this error:
> tsc

node_modules/modbus-serial/ModbusRTU.d.ts:2:22 - error TS7016: Could not find a declaration file for module './ports/testport'. '...../node_modules/modbus-serial/ports/testport.js' implicitly has an 'any' type.

2 import TestPort from './ports/testport';

8.0.6 is OK

yaacov, 2022-11-25:
Hi, thanks for the issue :fire:

Can you make a pull request fixing it ?

dbettoni, 2022-11-25:
I can try but I don't know when or time to do it,
for now the project works by setting the dependency to "modbus-serial": "8.0.6",

yaacov, 2022-11-25:
The issue looks related to https://github.com/yaacov/node-modbus-serial/pull/479
@chaos2171053 hi, can you take a look ?

@dbettoni can you add more information that will help understand the problem, link to your code ?
for example:
how you use the module in your project ?
how do you get the error ?

chaos2171053, 2022-11-29:
> On last version 8.0.7 my tsc build produce this error:
>
>
> node_modules/modbus-serial/ModbusRTU.d.ts:2:22 - error TS7016: Could not find a declaration file for module './ports/testport'. [...]

fix on this [pr](https://github.com/yaacov/node-modbus-serial/pull/484#issue-1467366277) .Can u provie more information to repoduce this bug?May be i can make this pr better.

yaacov, 2022-11-29:
#484 is merged @dbettoni can you check the fix ?

dbettoni, 2022-11-29:
I tried to rebuild my project by importing modbus-serial 8.0.7 but the problem hasn't been resolved.
see attached image
[image]

dbettoni, 2022-11-29:
my tsconfig.json:
[image]

yaacov, 2022-11-29:
@dbettoni hi, 8.0.7 does not have the fix, it will be in 8.0.8, can you test the code from github ?

chaos2171053, 2022-11-29:
> I tried to rebuild my project by importing modbus-serial 8.0.7 but the problem hasn't been resolved. see attached image [image]

Hi @dbettoni ,version 8.0.7 has not inclued the #484 ,you can use `npm install "https://github.com/yaacov/node-modbus-serial#master ---save` to use the latest code. 

[Install specific branch from github using Npm](https://stackoverflow.com/questions/39732397/install-specific-branch-from-github-using-npm)](https://stackoverflow.com/questions/39732397/install-specific-branch-from-github-using-npm)

@yaacov how about adding github ci scripts to publish the latest code to npm when pr is merged to the master or specific branch ?

yaacov, 2022-11-29:
> @yaacov how about adding github ci scripts to publish the latest code to npm when pr is merged to the master or specific branch ?

an action that publish on new release will be nice :-) 
but this fix will need the npm secret, I can look at a PR that assume an npm secret exist, and add the needed secret

dbettoni, 2022-11-29:
perfect, with the "modbus-serial" dependency: "git+https://github.com/yaacov/node-modbus-serial.git#master", everything is OK
Thanks for all

closed 2022-11-29 by dbettoni

yaacov, 2022-11-29:
8.0.8 is release :cake:

dbettoni, 2022-11-29:
OK Thanks