yaacov/node-modbus-serial #483: Connect to SEPLOS protocol

g992, 2022-11-28:
Friends, help me connect to the seplos bms protocol, I attach the file below. I am not an expert in RS485 communication, so it may not have been easy to connect. i used ```client.readCoils(2, 1), client.readHoldingRegisters(2, 1)```. Perhaps I do not understand anything and this library cannot simply be used with the SEPLOS protocol?
[SEPLOS BMS Communication Protocol_V2.0.pdf](https://github.com/yaacov/node-modbus-serial/files/10104534/SEPLOS.BMS.Communication.Protocol_V2.0.pdf)

yaacov, 2022-11-28:
Hi, thanks for the issue, I added a "help" label

g992, 2022-11-29:
How can i realize this protocol? [image]
How to set first command bytes?

tim-mills, 2022-11-29:
Did you get anywhere in this? Also trying to get readings from Seplos BMS but cannot make any sense of the Protocol docs.
Does it make any sense to anyone here? 
[SEPLOS BMS Communication Protocol_V2.0 (1).pdf](https://github.com/yaacov/node-modbus-serial/files/10112358/SEPLOS.BMS.Communication.Protocol_V2.0.1.pdf)

g992, 2022-12-05:
> Did you get anywhere in this? Also trying to get readings from Seplos BMS but cannot make any sense of the Protocol docs. Does it make any sense to anyone here? [SEPLOS BMS Communication [...]

Yes, i use simple node serial to communicate.

g992, 2022-12-05:
> Did you get anywhere in this? Also trying to get readings from Seplos BMS but cannot make any sense of the Protocol docs. Does it make any sense to anyone here? [SEPLOS BMS Communication [...]
u can check here: https://github.com/g992/modbus-seplos-bms-reader/tree/master