yaacov/node-modbus-serial #488: How Create ModbusRTUSlave?

teddy1565, 2023-01-16:
Hi, i tried create ModbusRTUSlave.
but the module looks like it's still in the works.

No available serialserver can use. it seems to share the servertcp_handler.js
I think there is something wrong with some interfaces.

for example:

An unknow error, blocking serialport:

ServerSerial Constructor:

```javascript
options = options || {};

const optionsWithBinding = {
    path: options.port || PORT,
    baudRate: options.baudrate || BAUDRATE,
    debug: options.debug || false,
    unitID: options.unitID || 255
};
if (options.binding) optionsWithBinding.binding = options.binding;

// create a serial server
modbus._server = new SerialPort(optionsWithBinding);
```
Make the following changes:

```javascript
 modbus._serverPath = new SerialPort({
      path: options?.path ?? PORT,
      baudRate: options?.baudRate ?? BAUDRATE,
      debug: options?.debug ?? false,
      unitID: options?.unitID ?? 255
});
```

I think it would be better to synchronize the option with the key of the SerialPort interface. 
Reduce misunderstandings under missing files.

another problem is serialport recive buffer with time difference.
so when i try get buffer, console tell me, i get this:
```
<Buffer 01 05 00 05 ff 00>
<Buffer 9c 3b>
```
but we expect can get:
```
<Buffer 01 05 00 05 ff 00 9c 3b>
```

so i thinik, we can use @serialport/parser-inter-byte-timeout, or other parser package, 
ensure parse only when we receive all packets.

Finally, I would like to ask whether it is possible to export serial server interface,
and have them referenced differently in the development environment
Because it unreasonable to reference serverTCP_handler in serverSerial

English is not my mother tongue, sorry for any offense

yaacov, 2023-01-16:
hi, thank you for your question, I added a "help wanted" label

you can try to look at the examples, you may find an answer in there:
https://github.com/yaacov/node-modbus-serial/tree/master/examples

teddy1565, 2023-01-16:
thanks, but examples doesn't have for serverSerial.
```javascript
const ModbusRTU = require("modbus-serial")
const serverRTU = new ModbusRTU.serverSerial(options)
```
The implementation of serverSerial and serverTCP should be different.
i think just solve the serialPort Parser & export interface.
the serverSerial can works great

teddy1565, 2023-01-16:
It can indeed pass in the CI test, but may need to consider physical factors.
In actual use, it is true that as long as it can solve the problem of receiving packets physically, it will be able to operate normally.

closed 2023-01-17 by yaacov