yaacov/node-modbus-serial #490: Need help with code in RTU format

Pavel-Demidyuk, 2023-01-19:
Hi!

I have an energy meter with Modbus support, and I need help reading its data using modbus-serial. 

In the manual there is following example:

Request code "01 04 00 00 00 02 71 CB"
01 - address
04 - read function
00 00 - internal register address
00 02 - data length, means 2 words 4 bytes of data
71 CB - CRC checksum

My code
```js
const ModbusRTU = require("modbus-serial");
const ID = 1;
const ADDRESS = 0x0000;
const HOST = "192.168.0.142"
const PORT = 8886;
const RATE = 9600

const client = new ModbusRTU();
client.connectTCP(HOST, {
    port: PORT,
    baudRate: RATE
});
client.setID(ID);
client.readInputRegisters(ADDRESS, 4)
    .then(console.log)
    .catch(console.log);
```

It return me nothing.

Please suggest what I'm doing wrong.

Thanks!

yaacov, 2023-01-21:
hi,
take a look at the examples directory, it may give you some ideas
https://github.com/yaacov/node-modbus-serial/blob/master/examples/simple.js

i see in the code snippet above that you don't wait for the connection to establish and you don't check for errors in the connection, that can be a good place to start debugging
i added a "help wanted" flag, in case someone else will have more ideas how to help here

HighWolf, 2023-04-12:
Hi,

If you are trying to connect to RTU (RX, TX wiring), you should use the `connectRTU` function, and your connection object will look something like this:
```ts
{
	path: '/dev/ttyAMA0',  // for Linux / COM# for windows
	id: 1,
	options: {
		baudRate: 9600,
		parity: 'none',
		dataBits: 8,
		stopBits: 1
	}
}
```

Where `/dev/ttyAMA0` / `COM4` represent the port path in your device (you'll need to check your device specifications).

Otherwise, if your connection is TCP (though RJ-45 connector / LAN), you won't need the baud rate and must consider that the default port for Modbus is 502.